raspberrypi/scratch #127: school live problem

heppg, 2015-02-02:
Hello, during the school session today, I got the following problem:
[image]
System is raspbian 2014-12-24 with updates, especially apt-get install nuscratch.

No clue on history. Just managed to get the screen photo.

Regards,
Gerhard

timrowledge, 2015-02-02:
Interesting; clearly a plain Pi has no recording hardware and you really ought not get to this part of the code.  I guess the proper thing would be to detect the lack of recording hardware and make sure the dialogue cannot open without it.

heppg, 2015-02-02:
Ah, now I understand. Can repeat this here when pressing 'record in sounds tab.
Problem in school is the lack of time to investigate a problem.

timrowledge, 2015-02-21:
The really annoying thing is that I can't replicate this, nor can I see how it is possible for the code to get to this state.
I *did* manage to discover that the unix sound recording low-level code is just plain nasty, amongst other things telling us that the recording sample rate is 0.0, which then causes a divide by 0 error. I've guarded against that ok but otherwise I'm puzzled.
I suppose it's just possible this is the version where Alex & I messed up the start script? That would result in the sound system being /dev/dsp by (bad) default) which could cause problems.

timrowledge, 2015-02-21:
Above mentioned change is in Scratch-tpr.238

closed 2015-02-21 by timrowledge