raspberrypi/scratch #129: animated gif not importing all frames.

heppg, 2015-02-23:
When importing an animated gif, only first view gets imported.
Expected: all frames get imported as separate costumes.

Example: wiki.scratch.mit.edu/wiki/Animated_GIF, there is a small animation on the page. Save this to disk, verify in a browser its animated and then import.

Works in scratch-windows and scratch.old
Current version: scratch-14-jan-15.

timrowledge, 2015-02-23:
Hmm. That's odd. I'm sure I made that work ages ago in order to import the two-costume gif for Gobo. Graphic file formats are *such* fun!

timrowledge, 2016-08-22:
Argh. By sheer coincidence when checking on Yet Another PNG Import Issue I spotted that the code had a ridiculous typo in handling animated gifs. With that sorted, it does actually import them, though I can make no guarantees that it does it well for many cases.

closed 2016-08-22 by timrowledge