raspberrypi/scratch #130: variable watcher inconsistent length truncation

heppg, 2015-03-14:
When having long string variable, the watcher seems to truncate displayed length to 64. This is reasonable (although old scratch tries to display even very long variables). But the watcher frame is still of a size as if the variable would be displayed completely.

Expected: The watcher frame width should be limited too.

Appended a simple script.
[image]

timrowledge, 2015-12-08:
It seems much more sensible to let the display actually show as much of the value as can fit, so that's what I've arranged.The limiting factor here was a slightly strange choice of default maximum width in the superclass of UpdatingScratchStringMorph; now we set our own value and use the ScratchFrameMorph>workpane extent x

timrowledge, 2015-12-08:
Fixed in Scratch-tpr.303

closed 2015-12-08 by timrowledge