raspberrypi/scratch #132: font error ?

heppg, 2015-05-11:
Hello,
kids in school have added text, and on 4 of 7 places they get the error message. The reason seems to be related with a scaled font. When deleting the costumes with scaled font, their projects could be saved, hopefully in good shape.
[image]
A damaged project is in heppg.de/Projekt_TJ_02.sb
I can't load it in windows nor in raspi-scratch.

timrowledge, 2015-05-11:
Can you find a squeakdebug.log file anywhere ? Do you know what font and size they were trying?

heppg, 2015-05-11:
Unfortunately, I only see the error message. When I load the damaged project linked above in legacy scratch, I get an error message. 
When I load it in raspi-scratch 2015-jan-14 (above mentioned raspi-scratch was an old one), I see the error message. 
The 'bad point' is in stage, background 2.
[image] The text is 'Spiel starten'. 
Deleting costumes with text (as seen in preview thumbnail) allowed to save the project.

heppg, 2015-05-11:
found a way to reproduce: created a sprite with "dejavu sans, 36". Saved as x00. Loaded again and found font described as 'a ScratchP..' in sprite editor. As I have seen these nested text in the project, I made following approach: saved project as x01. Loaded x01, saved as x02. Repeated till x05. When loading 'x05', I got the error message.

heppg, 2015-07-10:
Please see also
http://heppg.de/ikg/wordpress/?p=511

timrowledge, 2015-07-16:
This is an interesting example of how the naming of messages is important. If you use too-generic message names you find that sometimes the message is sent by unrelated code that expects a quite different value.
In this case, the 'ordinary' fonts have a method named 'name' that returns the font's name - which seems extremely sensible. It is sent as part of writing out a text box to a scratch project; we just write the text box's font's name. Very reasonable. Unfortunately, 'name' is also implemented as a method in the root Object class as part of making the UI for object inspectors and the default is just 'self printString'. In Smalltalk  the printString message is used to ask an object to describe itself in text form, so when a pango outline font is written to a Scratch project the string used is 'a ScratchPangoFont(...' etc.

The fix for this is just to implement #name for ScratchPangoFont to return the font name and not the long winded printString.

Fixed in Scratch-tpr.252

closed 2015-07-16 by timrowledge