raspberrypi/scratch #133: exporting background to write protected directory

heppg, 2015-06-18:
Export a background to a write protected directory gives the following error screen:
[image]
Expected: should give an kid-compatible error message.

timrowledge, 2015-06-18:
It certainly should!

I'll get back to working on stuff like this soon; currently very busy with making the system a lot faster.

heppg, 2015-06-19:
Hello Tim,
this ticket is low prio. The 'pangofont'-problem reported earlier is
really high prio, as it does not allow to use text in sprite/backgroud.
Regards,
Gerhard

timrowledge, 2015-07-16:
I'd like to make it so you can't choose a bad filename but that isn't looking too easy right now.

timrowledge, 2016-10-31:
I've wrapped the image media exporting in an error handler so at least it will provide a notification rather than crash.
Fixed in NuScratch-tpr.376

closed 2016-10-31 by timrowledge