raspberrypi/scratch #134: NuScratch140115 crashes on raspberry pi when using sound and remote sensors

thies-How, 2015-06-23:
When I enable and use a tcp remote sensor connection and play a sound (the default drum) scratch crashes with the following message:

> sound_Start(default)
> I/O possible

This occurs in the current raspbian image (201505-05-raspbian-wheezy.zip SHA-1: cb799af077930ff7cbcfaa251b4c6e25b11483de) and the current NOOBS raspbian install (date 2015-06-23). I have tested using a clean install and also fully upgraded (apt-get upgrade and rpi-update) with same result.

Using scratch.old (which starts scratch using the Scratch20131203.image) it works without crashing.

How to see the crash:

1. Install a clean raspbian on a raspberry pi 2.
2. Boot to LXDE Desktop
3. Run 'scratch' from a lxterminal (so you can see the error message). Scratch will write following on the terminal:
>Executing: /usr/bin/squeak-stack -memory 128M -plugins /usr/lib/squeak/4.10.2-2793/ -vm-sound-alsa /usr/share/scratch/NuScratch140115.image

4. Enable remote sensor connection:
    * Go to the light blue ”Sensing” panel
    * Right click [sensor value]
    * Choose ”enable remote sensor connections”
5. Download [minscratch.py](http://pastebin.com/fFScsWHj) and run it from another terminal. It should print out ”Connected!”. The program connects to scratch and updates sensor ”x” with the values 0.0 to 100.0 in a loop. You can cancel it with ctrl-c
6. Play the drum by clicking `play drum '48' for '0.2' beats` in the pink ”Sound” panel. If it does not crash immediately, try clicking a couple of more times. Make sure you can play and hear sound in your setup.

Scratch will crash with the output:
>sound_Start(default)
>I/O possible

What I noticed:

1. Running the above example but using scratch.old instead of scratch works!
This writes the following on the terminal:
> Executing: /usr/lib/squeak/4.10.2-2793/squeakvm -encoding UTF-8 -vm-display-x11 -xshm -plugins /usr/lib/scratch/plugins/:/usr/lib/squeak/4.10.2-2793/ -vm-sound-alsa /usr/share/scratch/Scratch20131203.image

2. Playing sounds without remote sensors works fine
3. Using remote sensors without sounds works fine as well
4. Enabling remote sensors without connecting works fine
5. Enabling remote sensors and connecting without updating sensors through the socket connection works fine.
6. Enabling sensors, playing a sound, then connecting crashes immediately
7. It makes no difference using the sampled sounds compared to the built in drum sounds.
8. Scratch will not crash, if you start scratch with another sound plugin ”-vm-sound-null” for example:
> /usr/bin/squeak-stack -memory 128M -plugins /usr/lib/squeak/4.10.2-2793/ -vm-sound-null /usr/share/scratch/NuScratch140115.image

9. I googled for ”I/O possible” message and think there is a relation to `SIG_IO` used by ALSA to synchronize its sound buffer writes (I guess). I tried to download the squeak-vm source via apt-get source squeak-vm, play around with the plugin code for the -vm-sound-alsa plugin and build it. All my attempts ended in either no sound or a crash at the exact same point described above. I was not able to find if the bug resides in that section of the code, as there seem to be multiple threads running. It seems like scratch would crash during the execution of `sound_Stop()` in unix/vm-sound-ALSA/sqUnixSoundALSA.c

Any hints are greatly appreciated.

Keep up the good work.

Best regards

timrowledge, 2015-06-23:
Thank you for a clear and detailed explanation. That will help a great deal and I wish all problem reports could be so clear!

We've had a lot of problems relating to ALSA. I'd love to find a better system to use.

timrowledge, 2015-07-16:
Yup, this is ALSA's problem. We had lots of fun tracking down just this issue in developing the new CogVM (coming soon to a Pi near you) and the alas (hah! autocorrect gets it right!) code explicitly sets the sigio handler to SIG_DFL when the sound player stops at the end of a sound. The default response to sigio is to terminate the process! Which is another level of insanity, but never mind.
The correct fix (making alsa do the Right Thing with chaining handlers) is being considered at Pi Towers and should go into Raspbian soon.

closed 2015-07-16 by timrowledge