raspberrypi/scratch #135: Touch invisible objects

heppg, 2015-07-07:
Enhancement request: kids in school found following script 
[image]
The idea is to click on a sprite, which then disappears. It is unexpected that the sprite reacts to mouse touching even when hidden.
This is same behaviour on win-scratch-1.4 and current scratch on raspbian, so it is no fault.
A more intuitive behaviour would be: do not react on mouse events when hidden.

We then found the 'when Object clicked' more useful

timrowledge, 2015-07-07:
Hmph. I'd consider that a bug actually.

Of course, the problem is that some people may be relying upon it working that way. But I'm still seeing it as a bug.

seanmcmanus, 2016-04-05:
This behaviour is consistent with how Scratch handles invisible sprites touching colours (which is a very useful capability), so I'm not sure it qualifies as a bug. But Scratch is inconsistent on this: Invisible sprites can't detect if they're touching other sprites.