raspberrypi/scratch #136: Remote sensor netword, scratch send error when utf-8 chars are in variable name

heppg, 2015-07-09:
When variable names contain utf-8 chars, e.g. "Münzen", then the size part of the message and the number of bytes sent do not match. According to the spec in http://wiki.scratch.mit.edu/wiki/Remote_Sensors_Protocol, the length is in bytes.
[image]
The output in scratchClient is:
`2015-07-09 19:15:02,519 - __main__ - 342 - DEBUG - ScratchListener thread started
2015-07-09 19:15:02,526 - __main__ - 371 - DEBUG - received 00 00 00 19 73 65 6e 73 6f 72 2d 75 70 64 61 74 65 20 22 4d c3 bc 6e 7a 65 6e 22 20 30 20
2015-07-09 19:15:02,531 - __main__ - 426 - DEBUG - data received from scratch-Length: 25, Data: sensor-update "MÃ¼nzen" 0
2015-07-09 19:15:02,539 - __main__ - 463 - DEBUG - sensor-update rcvd sensor-update "MÃ¼nzen" 0
2015-07-09 19:15:02,547 - __main__ - 479 - INFO - sensor-update: MÃ¼nzen, 0
2015-07-09 19:15:26,361 - __main__ - 371 - DEBUG - received 00 00 00 19 73 65 6e 73 6f 72 2d 75 70 64 61 74 65 20 22 4d c3 bc 6e 7a 65 6e 22 20 31 20
2015-07-09 19:15:26,363 - __main__ - 396 - ERROR - fatal: first two bytes of message are not zero, discard data till zeros found
2015-07-09 19:15:26,371 - __main__ - 404 - ERROR - discarded 1 bytes
2015-07-09 19:15:26,375 - __main__ - 426 - DEBUG - data received from scratch-Length: 25, Data: sensor-update "MÃ¼nzen" 1
2015-07-09 19:15:26,381 - __main__ - 463 - DEBUG - sensor-update rcvd sensor-update "MÃ¼nzen" 1
2015-07-09 19:15:26,391 - __main__ - 479 - INFO - sensor-update: MÃ¼nzen, 1
2015-07-09 19:15:36,402 - __main__ - 371 - DEBUG - received 00 00 00 19 73 65 6e 73 6f 72 2d 75 70 64 61 74 65 20 22 4d c3 bc 6e 7a 65 6e 22 20 32 20
2015-07-09 19:15:36,404 - __main__ - 396 - ERROR - fatal: first two bytes of message are not zero, discard data till zeros found
2015-07-09 19:15:36,412 - __main__ - 404 - ERROR - discarded 1 bytes
2015-07-09 19:15:36,414 - __main__ - 426 - DEBUG - data received from scratch-Length: 25, Data: sensor-update "MÃ¼nzen" 2
`
Current situation, the first block of bytes received at 2015-07-09 19:15:02,526 is
len: 00 00 00 19 
data: 73 65 6e 73 6f 72 2d 75 70 64 61 74 65 20 22 4d c3 bc 6e 7a 65 6e 22 20 30 20
Length is x19 ==25, but bytes received are 26
As scratchClient puts as much bytes given in the len-Information to processing, in this case one byte remains in buffer, which is concat with next data record and produces an incredible large length information. The log output shows already a recovery strategy in my scratchClient, which in case of len information >> 0xffff looks for two consecutive zero bytes in data stream. This is the log message at 2015-07-09 19:15:26,363. 

expected, the len block for the data block above should be 00 00 00 1A

timrowledge, 2015-07-16:
Nice catch. Thorough bug report makes it nice and simple to find the cause, which was a trivial dumb-thumbs where
```msg := outgoingString.
broadcast msg size.
broadcast msg convertToUTF8```
should be
```msg := outgoingString convertToUTF8.
broadcast msg size.
broadcast msg```

timrowledge, 2015-07-16:
Fixed in Scratch-tpr.251

closed 2015-07-16 by timrowledge