raspberrypi/scratch #138: bad header error

brianlean, 2015-07-16:
Running a scratch on Pi camp and the girls in camp keep getting the same error.  Related to adding backgrounds or sprites to a project.  
[image]

timrowledge, 2015-07-16:
Please upload copies of any file you can possibly imagine might help track this down (your project ,the sprites, etc), plus any recipe you can come up with to replicate it. You should find a file called 'SqueakDebug.log' or similar in your user directory that starts with something along the lines of 
"Error: bad header
date and time of error

VM: unix - Smalltalk
Image: Squeak 4,5" etc
Send that too since it provides a backtrace of how you got there.

brianlean, 2015-07-16:
Holly

Good news.  The Raspberry Pi folks are working a new version of Scratch.  I
pulled the latest code and will try to bring that by today and I have this
bug logged with them.  If it happens again try to capture as much info as
possible including getting a copy of the project and the text in the error.

Thanks

Brian S. Moran
The Boolean Girl Project
www.booleangirl.org
703.608.1941

brianlean, 2015-07-16:
Tim

I absolutely will pull together more info on this.

Brian S. Moran
The Boolean Girl Project
www.booleangirl.org
703.608.1941

timrowledge, 2015-08-01:
Brian, any progress on this?

brianlean, 2015-08-01:
Tim

sorry - I was not able to get anything useful off the Pi that had the error
2 weeks ago.  Good news is one of our camps had the same issue at the end
of the week.  I'll get the Pi back tomorrow and hope to have something for
you to work with.

Thanks

Brian S. Moran
The Boolean Girl Project
www.booleangirl.org
703.608.1941

brianlean, 2015-08-04:
Tim

I got the Pi's back and I cannot find the debug file you are looking for.
I'm looking in the user directory home/pi can't find it.  Any ideas?

I can tell you what happened but the file the girl was building no longer
exists.  She was using the paint editor to create a new sprite.

I'll keep trying to reproduce it.  Let me know about the debug log file.

Brian S. Moran
The Boolean Girl Project
www.booleangirl.org
703.608.1941

timrowledge, 2015-08-04:
On 04-08-2015, at 3:23 AM, BrianM <notifications@github.com> wrote:
> I got the Pi's back and I cannot find the debug file you are looking for.
> I'm looking in the user directory home/pi can't find it. Any ideas?

Ah, of course, forgot to mention that part. It is a bit of a pain but that�s just how unix likes to make it - it would have ended up in /usr/share/scratch/SqueakDebug.log

Someday I�ll have a moment to change where it gets written and make it somewhere easier to find/mail to gihub

tim
--
tim Rowledge; tim@rowledge.org; http://www.rowledge.org/tim
..... REALITY.SYS Corrupted - Unable to recover Universe

brianlean, 2015-08-04:
Ok - will dig it out tonight.

Brian S. Moran
The Boolean Girl Project
www.booleangirl.org
703.608.1941

brianlean, 2015-08-05:
Tim

Bad news - no log file

Good news - I recreated the issue.  Still no log file but the .img file
from the Pi is at .
https://www.dropbox.com/sh/zb1mdkcx3camzxl/AADMQgg-VR9A5MQB-oRI8aZYa?dl=0

It is really interesting, the file is now 0 bytes so I can't send it.  It
is on the image, on the desktop of the pi users.

Let me know what else I can do to help.

Brian S. Moran
The Boolean Girl Project
www.booleangirl.org
703.608.1941

timrowledge, 2015-08-05:
How weird. Well I'll look into it sometime soon. Don't suppose you remember how you triggered it? A recipe can be very useful.

brianlean, 2015-08-05:
Tim

Yep - It was paint.  I kept painting and adding sprites.  Building small
scripts for each then I saved, closed, reopened and there was the error.

One thing I noticed is I must have been pushing the limit on memory.  A
temp file would open on the desktop on each save.  The project file was on
the desktop.

Brian

Brian S. Moran
The Boolean Girl Project
www.booleangirl.org
703.608.1941