raspberrypi/scratch #139: scratch-2015-07-18 to scratch-win-1.4 compatibility

heppg, 2015-07-24:
scratch to windows character display check.
created on rpi a project with elements containing umlaut (utf-8) chars then transferred to windows and opened with scratch-win-1.4. Result are:
- text in costume: displayed correctly, but not in costume editor
- variable name; not displayed correctly
- broadcast event name: not displayed correctly
- project comments: not correct
- project author: not correct
- sprite name: not correct

timrowledge, 2015-08-01:
Could you upload a project that illustrates this?

heppg, 2015-08-02:
Here screenshots:
Project created with raspbian, 2015-07-18 scratch. Transferred to windows, opened with legacy scratch 1.4, here on raspbian
[image]
here on windows
[image]

heppg, 2015-08-02:
The project is on www.heppg.de/Muenzen.sb

timrowledge, 2016-11-29:
Argh.

This is a side-effect of how the very ancient Squeak system used for the legacy Scratch was based around the macRoman character set; the newer Squeak uses latin-1. For a small number of characters this seems to cause a bad mapping after all the conversions between macRoman, utf8, utf32 and latin-1 have been passed though.

Adding in some macRoman stuff might be possible but to be honest the more sensible thing to try would be to use NuScratch on your Windows and Mac machines. All you need to do is download the new 'cog spur' VM packages from www.squeak.org and run the same .image file as you have on the Pi. You can find that in /usr/share/scratch - just copy the NuScratchXXXXXX.image file with the latest XXXXXX date number to your pc/mac and set up whatever aliases needed to make it be run with the new VM.

heppg, 2016-11-29:
Quite old ticket. Should be retested with current nuscratch on pi. Will
do this next days.
The impact of this ticket is unfortunately quite high. It affects the
school usecase "carry your project file home, continue with scratch 1.4
you can download from scratch web site and bring back your results next
week".
This requires that file formats and code pages are the same between
raspberry and window worlds. Only 1-3 kids out from 26 use a pi at home.

- kids in general are not able to perform complicated install procedures
on win computers, possibly needing admin rights
- file formats should be kept stable.

timrowledge, 2016-11-29:
Unless I missed something I've already tested with effectively the latest Pi versions, so don't spend too much time on it.

Good point about the kids installing stuff. I think I can fix that by assembling some simple downloads of the new system that are no more trouble than the old one.
It's probably a good idea to push to move past the very, very, old VMs provided from MIT anyway since they're >10 years old and we know there are issue with older VMs on recent OS updates.

I agree that file formats should be kept stable as much as possible but sometimes it just costs more than can be afforded.