raspberrypi/scratch #142: scratch-2015-07-18 piglow causes exit

heppg, 2015-08-01:
pi2
wiringpi installed as described on http://wiringpi.com/download-and-install/
i2c enabled by raspi-config
Started scratch
loaded gpio-PiGlow.sb

result is: scratch exits without notice.
debug log is containing multiple sessions.

Error: External module not found
1 August 2015 8:12:24.597 am

VM: unix - Smalltalk
Image: Squeak4.5 [latest update: #15047]

SecurityManager state:
Restricted: false
FileAccess: true
SocketAccess: true
Working Dir /home/pi/fastScratch
Trusted Dir /home/pi/fastScratch/secure
Untrusted Dir /home/pi/fastScratch/My Squeak

ScratchWiringPiInterface(Object)>>error:
	Receiver: a ScratchWiringPiInterface
	Arguments and temporary variables: 
		t1: 	'External module not found'
	Receiver's instance variables: 
a ScratchWiringPiInterface
ScratchWiringPiInterface(Object)>>externalCallFailed
	Receiver: a ScratchWiringPiInterface
	Arguments and temporary variables: 
		t1: 	15
	Receiver's instance variables: 
a ScratchWiringPiInterface
ScratchWiringPiInterface>>wiringPiSetupGpio
	Receiver: a ScratchWiringPiInterface
	Arguments and temporary variables: 

	Receiver's instance variables: 
a ScratchWiringPiInterface
ScratchWiringPiInterface class>>uniqueInstance
	Receiver: ScratchWiringPiInterface
	Arguments and temporary variables: 
		t1: 	a ScratchWiringPiInterface
	Receiver's instance variables: 
		superclass: 	Object
		methodDict: 	a MethodDictionary(#analogRead:->(ScratchWiringPiInterface>>#analog...etc...
		format: 	0
		instanceVariables: 	nil
		organization: 	('pin handling' analogRead: analogWrite:value: digitalRead: digit...etc...
		subclasses: 	nil
		name: 	#ScratchWiringPiInterface
		classPool: 	a Dictionary(#UniqueInstance->nil )
		sharedPools: 	an OrderedCollection(WiringPiGPIOConstants)
		environment: 	Smalltalk
		category: 	#'Scratch-Plugins'

GPIOAccessorWiringPi>>initialize
	Receiver: a GPIOAccessorWiringPi
	Arguments and temporary variables: 

	Receiver's instance variables: 
		library: 	nil

GPIOAccessorWiringPi class(Behavior)>>new
	Receiver: GPIOAccessorWiringPi
	Arguments and temporary variables: 

	Receiver's instance variables: 
[... 416 lines cut ...]
MenuItemMorph(Morph)>>handleEvent:
MorphicEventDispatcher>>dispatchDefault:with:
MorphicEventDispatcher>>dispatchEvent:with:
MenuItemMorph(Morph)>>processEvent:using:
MorphicEventDispatcher>>dispatchDefault:with:
MorphicEventDispatcher>>dispatchEvent:with:
ScratchMenuMorph(Morph)>>processEvent:using:
ScratchMenuMorph(Morph)>>processEvent:
ScratchMenuMorph(MenuMorph)>>handleFocusEvent:
[] in HandMorph>>sendFocusEvent:to:clear:
BlockClosure>>on:do:
PasteUpMorph>>becomeActiveDuring:
HandMorph>>sendFocusEvent:to:clear:
HandMorph>>sendEvent:focus:clear:
HandMorph>>sendMouseEvent:
HandMorph>>handleEvent:
HandMorph>>processEvents
[] in WorldState>>doOneCycleNowFor:
Array(SequenceableCollection)>>do:
WorldState>>handsDo:
WorldState>>doOneCycleNowFor:
WorldState>>doOneCycleFor:
PasteUpMorph>>doOneCycle
[] in MorphicProject>>spawnNewProcess
[] in BlockClosure>>newProcess

timrowledge, 2015-08-01:
This is actually a 'feature' of wiringPi that needs to be changed; if you try to connect again to someinth that is already setup it dumps the process instead of returning an error. Not really a good idea.

timrowledge, 2015-11-13:
I've added some code in the initialisation of i2c devices that *might* reduce this issue's impact.It's in the November 15 release candidate so let's see what happens.

timrowledge, 2016-03-29:
see also #183

timrowledge, 2016-04-29:
Closed along with #183

closed 2016-04-29 by timrowledge