raspberrypi/scratch #145: scratch-2015-07-18 naming of sensor values

heppg, 2015-08-24:
Proposal:
current implementation: pin11on pin18pwm10 pin24low
proposed naming:  gpio11on gpio18pwm10 gpio24low

In current world, there are header pins and gpio names. During a presentation of Clive Beale in Amsterdam it got obvious that audience was confused by an image where pin numbers and gpio numbers have been used. And that pin-names in scratch mean gpio numbers which need to be matched to pin numbers...
So it would be more clear to use gpio-naming.

timrowledge, 2015-08-24:
On 24-08-2015, at 12:16 PM, heppg <notifications@github.com> wrote:

> Proposal:
> current implementation: pin11on pin18pwm10 pin24low
> proposed naming: gpio11on gpio18pwm10 gpio24low

A good point. I�ll think about that.

tim
--
tim Rowledge; tim@rowledge.org; http://www.rowledge.org/tim
Useful Latin Phrases:- Aio, quantitas magna frumentorum est. = Yes, that is a very large amount of corn.

clivebeale, 2015-08-24:
Been thinking about this also and I agree  -- ties in with naming conventions in Python library and makes a distinction when you say eg "Pin 6 is a ground". (Plus will keep engineers happy :p)

timrowledge, 2015-09-18:
Implemented for sept'15 release

closed 2015-09-18 by timrowledge