raspberrypi/scratch #147: scratch-15092015, getip wrong result

heppg, 2015-09-16:
My local raspberry has 192.168.2.108 (ifconfig, and can ping that from other computers with this address)
Result of getip is 127.0.1.1, which i snot what I expect.

timrowledge, 2015-09-16:
This is very odd. Under gdb we find the correct local host name but the code uses nameToAddr() to convert that to an IP and we get 127.0.1.1 out of it. I can't even find any trace of documention for this function with google.

Asking around...

timrowledge, 2015-09-17:
Ah, it's the way Debian (and thus Raspbian) sets up the /etc/hosts data. I suspect it changed some time not too long ago because I fairly sure this used to work was expected or we would have had many, many complaints about not being able to use any of the mesh/socket external gpio programs.

The quick solution is to
`sudo nano /etc/hosts`
and change the (probably) last line from something like
127.0.1.1      mymachinename
to
192.168.1.99 mymachinename
with whatever appropriate numbers and names.

timrowledge, 2015-09-17:
Simon, dumping this on you because you're just so lucky :-) And you may know to whom you should  pass it.

heppg, 2015-09-17:
from http://stackoverflow.com/questions/212528/get-the-ip-address-of-the-machine

#include <stdio.h>      
#include <sys/types.h>
#include <ifaddrs.h>
#include <netinet/in.h> 
#include <string.h> 
#include <arpa/inet.h>

int main (int argc, const char * argv[]) {
    struct ifaddrs * ifAddrStruct=NULL;
    struct ifaddrs * ifa=NULL;
    void * tmpAddrPtr=NULL;

    getifaddrs(&ifAddrStruct);

    for (ifa = ifAddrStruct; ifa != NULL; ifa = ifa->ifa_next) {
        if (!ifa->ifa_addr) {
            continue;
        }
        if (ifa->ifa_addr->sa_family == AF_INET) { // check it is IP4
            // is a valid IP4 Address
            tmpAddrPtr=&((struct sockaddr_in *)ifa->ifa_addr)->sin_addr;
            char addressBuffer[INET_ADDRSTRLEN];
            inet_ntop(AF_INET, tmpAddrPtr, addressBuffer, INET_ADDRSTRLEN);
            printf("%s IP Address %s\n", ifa->ifa_name, addressBuffer); 
        } else if (ifa->ifa_addr->sa_family == AF_INET6) { // check it is IP6
            // is a valid IP6 Address
            tmpAddrPtr=&((struct sockaddr_in6 *)ifa->ifa_addr)->sin6_addr;
            char addressBuffer[INET6_ADDRSTRLEN];
            inet_ntop(AF_INET6, tmpAddrPtr, addressBuffer, INET6_ADDRSTRLEN);
            printf("%s IP Address %s\n", ifa->ifa_name, addressBuffer); 
        } 
    }
    if (ifAddrStruct!=NULL) freeifaddrs(ifAddrStruct);
    return 0;
}

spl237, 2015-09-18:
I am assuming getip is a scratch function to determine the IP address of the host computer? Reading this from /etc/hosts is the wrong thing to do - this just provides address mappings before DNS is available, and the host machine's address is likely to change once it has established a connection to DNS. There is also the question of which IP address are you asking for if you have, for example, both an Ethernet cable and one or more USB wifi dongles connected?

heppg, 2015-09-18:
Think the case that someone has multiple adapters connected is quite rare. People having this setup could be considered 'experts': able to understand 'ifconfig' command line tool.
Most people will use one adapter.
Think it is good enough to return the first address found, not equal to 127.0.0.1

timrowledge, 2015-09-18:
The intent of the getIP thing is to give the user an address they can enter on another scratch system so as to join the mesh network. So I guess it has to be a 'good' IP number for the hosting device, whatever that might mean. 127.0.1.1 is pretty much never going to be a useful answer for this.

It's not a specifically Scratch thing, nor even new code; I think the C code implementing the lowest levels in the VM has been stable for maybe ten years now (also known as 'nobody has bothered to improve anything in ages') because it does what seems right on other machines & unixes. I'm puzzled as to why any OS would answer the 127 value when there are clearly better results any time there is any sort of network connection. 

There must surely be a standard api for this? I mean, it's got to be a really common requirement

heppg, 2015-09-18:
perhaps you want to have a look into #144 -problem. The font problem covers local error ?

timrowledge, 2015-09-21:
We've had a fair bit of discussion about this on the Squeak forums and it appears that changes in debian/ubuntu/etc are the major cause of this problem. I am told that it was never a reliable approach despite our general experience being fine.

I don't think we can do much until the next release and will need to document the problem for continuing users of the mesh networks; one way to dodge the issue is to manually edit the /etc/hosts file to add the local host name and IP, though anyone using dhcp may have issues. Alternatively, remind people that they can find their machine IP by other methods like ifconfig and enter that into the dialogue on the connecting machines.

For a future release we can probably use a brief net transaction to establish a socket and thereby get the local IP address established; A code snippet like - 
| foo bar|
foo := Socket newTCP.
foo setPeer: #[8 8 8 8] port: 80.
bar := foo localAddress.
foo close.
bar
works as an example. However, 
- it can’t work (presumably) if the machine is not on an externally connected network, which is quite plausible in many schools in many countries
- [8 8 8 8] is the sort of magic number nonsense I hate to see.

timrowledge, 2015-11-05:
I've found a (possibly) better solution using getifaddrs() in the relevant socket primitive. So far it appears to work correctly.

timrowledge, 2015-11-05:
Added (hopeful) support for wifi addresses; since there is no good way to offer multiple IP numbers to the user I'm accepting the first found plausible answer which then depends upon whatever the local implementation of ifgetaddrs() does.

timrowledge, 2015-11-05:
Improved code should go into the november release

closed 2015-11-05 by timrowledge