raspberrypi/scratch #149: jessie and scratch integration

heppg, 2015-09-29:
The 'help' menu button opens libreoffice in edit mode for the html help pages.

Expected: should be opened in browser.

timrowledge, 2015-09-30:
Ah, this old problem returns; for some time it would open the very old web-browser. We use the unix 'xdg-open' (or something like that) api to ask the system to handle an html file and it does whatever it thinks best. I guess it isn't making a very good choice right now...

timrowledge, 2015-09-30:
Simon - another one I think you need to know about; looks like some config doohickey that xdg-open looks at is mixed up?

spl237, 2015-09-30:
xdg-open just calls the system file manager, so it is the file associations in pcmanfm that need to be fixed - libreoffice seems to grab a lot of them when it is installed. The easy fix is to open the File Manager, navigate to a .html file (such as /usr/share/scratch/Help/index.html), right-click and choose Epiphany to open it, ticking the box for "always use this application to open files of this type". That creates the required association. 

We'll look at including a predefined association with .html files in the next release.

timrowledge, 2016-01-11:
Seems to have been solved for the november release at least.

closed 2016-01-11 by timrowledge