raspberrypi/scratch #150: scratch-15092015, sound 'pop'

heppg, 2015-10-05:
as reported by by mob-i-l » 05 Oct 2015, 01:21 
https://www.raspberrypi.org/forums/viewtopic.php?p=824627#p824627

Sound effects/pop is not reliably played. In a loop,  forever: play sound 'pop' until done, the sound is not played. Other sounds work. This is reproducible in my 'jessie' raspbian environment.

timrowledge, 2015-10-06:
That's curious...
To add to the fun my primary development Pi decided to stop sending noises to the audio out socket whilst testing this. Rebooting didn't change anything. Sigh.
A secondary testing Pi makes noises ok and 'meow' etc work ok but yes,, 'pop' just makes no sound - except that the sound system debug output says it does!
This may take a little puzzling over!

timrowledge, 2015-10-06:
(later) Yup, there's something about the amount of buffer data; if I stop the sound player from terminating the sound playing when there is no more data (err, that reads badly - the sound player loop fills in 0 data when there is no more 'real' data. We can detect that and suspend the player process) then the pop plays ok.

timrowledge, 2015-11-11:
Somewhere buried deep in the SoundPlayer code there is a mistake that is proving very difficult to find. I actually managed to get values set up so the pop would reliably play - and then it got out of sorts again.

timrowledge, 2016-01-11:
So far this is eluding us. The pop sounds ok if we have continuous playing on - but that leaves terrible pwm jitter hiss behind - or reverb on, which isn't so nice either. Works the same on iMac too, so at least it should be image fixable eventualy.

timrowledge, 2016-08-25:
Latest testing with the build intends for sept '16 release, built on the production Squeak 5.1 release *appears* to play the pop ok. Along with recent audio firmware and some Squeak work, I think this is fixed.

timrowledge, 2016-09-13:
Related to #218