raspberrypi/scratch #153: Allow frequency of gpio PWM control

timrowledge, 2015-10-12:
Looks like we can at least roughly set the wpm frequency, which is apparently useful for motors.
See https://www.raspberrypi.org/forums/viewtopic.php?p=828302#p828302 & wiringPi/softPwm.c

timrowledge, 2015-12-19:
There isn't really a clean interface to wiringPi's function to set up wpm frequency, which is a bit of a pain. It looks like maybe we can directly call softPwmCreate (origPin, initialValue, range) ; but we'd also have to carefully stop any prior wpm or tone on that pin. The practical range is limited since wiringPi works with a minimum pulse width of 100microsec and 'range' is used to set how many 100usec periods make a full cycle. Thus setting range to 100 gives us 100Hz, range = 10 gives us 1000Hz, range = 1000 gives us 100Hz and so on. Clearly this limits the precision of the values as well in an inverse manner.

timrowledge, 2016-01-05:
OK, added an API to allow setting a frequency for the softPWM support. There are a lot of practical restrictions though; see above. Unfortunately this really prevents a good solution to #157 as well.
Need to test this with a silly-scope before release.

timrowledge, 2016-02-08:
Released in january release

closed 2016-02-08 by timrowledge

NoSheds, 2017-10-23:
On 5th Jan 2016 I see that you've said you've added an API to adjust the PWM frequency, but despite much searching, I can't find out how, or any documentation.  Could you give me an idea of how to do that please?

timrowledge, 2017-10-23:
OK, this is complicated by the fact that we changed from wiringPi to pigpio sometime after this was 'fixed' - pigpio has a different idea of what setting the pwm frequency means.

What are you actually wanting to achieve here? I might be able to point to a solution within the current system.

NoSheds, 2017-10-23:
I have used Simon Walters' (@cymplecy)'s scratch add on to use PWM to control a model train using a driver chip to PWM the output of the train's main's controller.  The controller produces 12V DC, which I PWM to control the train speed.  Somehow, I "blew" the driver chip so I talked to a friend who has an electronics business.  He said, contrary to the theory of running the frequency very slow for motors described above, what is required (at least for me) is to run it at as high a frequency as is possible, otherwise the power supply & components take a hammering.  I double-checked this with another electronics bod who confirmed.
Using Simon's code, I can get the frequency up to about 8KHz.  I was wondering if I could better this with the RPI's version of the Scratch add-on, especially if it could do hardware PWM.

timrowledge, 2017-10-23:
Ah, right. Well this is way outside any scenario I was ever asked to consider and right now there isn't anything within the GPIOServer code to handle it. It's interesting enough that if you're willing to work on some of the problem then I'm willing to spend some time to add it.

First things first, according to joan at http://abyz.me.uk/rpi/pigpio/pdif2.html#hardware_PWM the absolute maximum frequency is 125MHz, which I have to say seems insanely high (lower frequencies are available at all good pigpio libraries, as they say) and you'd need to find out if your driver circuit could even begin to handle it. I imagine that getting a signal like down a wire for any distance might be an issue too - consult your Local Friendly Electronics Nerd.

Once a sensible frequency is selected you'd need to try out some usage with the `pigs` commandline stuff - http://abyz.me.uk/rpi/pigpio/pigs.html in particular I think http://abyz.me.uk/rpi/pigpio/pigs.html#HC is the most important. That ought to let you establish that you can wire up the right bits the right way and get some control over them. If you have access to an oscilloscope it would help and might avoid blowing up a driver chip or Pi.

NoSheds, 2017-10-23:
The driver is a Pololu Motor Driver 1A DRV8801 and can run up to 50KHz.  I think last time I talked to someone about this they said around 30KHz was OK.  Anything in the range 20Hz to 22KHz may be heard as the motor pulses in the audible range.  Just to complicate things, I use some servos too to change points, so the PWM can't stop the servos from working.  I will talk to my electronics nerd about if the signal can travel the required distance.  I will give the "pigs" a go and see what's what.  Unfortunately, I don't have a 'scope :-(  Is this the right place to discuss this, or should we do it in the forums somewhere, or take it to email?

timrowledge, 2017-10-23:
That's much less scary than MHZ!

You can easily manage 40KHz with software pwm, without messing up servo control. All you need to do is start pigpiod with 
`sudo pigpiod -s1`
so that the sampling time is set as small as it can manage; the downside is that it will use about 3% of cpu on a Pi3 (so likely actually 10% on a single core) but controlling a model train or ten ought not cause too much hassle.
Check carefully for yourself but I think that you can test it out with 'pigs' by
(lets assume you are using pin bcm 17, check https://pinout.xyz/pinout/pin11_gpio17)
`sudo killall pigpiod` (to kill any current version)
`sudo pigpiod -s1`
`pigs pfg 17` (likely to return default 800)
`pigs pfs 17 40000`
Then `pigs p 17 X` will set the pwm duty cycle to x/255, so X=0 is off, X=128 is half, etc.

I did a built-in driver for Scratch to run the pololu8835 which is very similar (and includes the direction pin like your 8801, don't forget to set that correctly or you may get odd effects) but it assumes 8KHz is ok and a 0-100 duty range.

And as for a proper place to discuss this - well, who's going to care?