raspberrypi/scratch #155: PiFace inputs need pulp set to reduce jitter

timrowledge, 2015-10-13:
https://github.com/raspberrypi/documentation/issues/262 suggests a solution to the PiFace input jittering; looks like wiringPi provides a simple enough way to set the input pullups - see wiringPi/examples/PiFace/buttons.c

stevensmi, 2015-10-19:
I see this problem as well. The pullups should either be activated by default or control provided so this can be turned on.

I would be quite happy to submit a patch. However as far as I can seethe repository provides no source code for the raspberrypi version of scratch. It would be very helpful to build a community if the source code was easily available. Committing the .tgz in GIT seems very odd!

timrowledge, 2015-10-19:
Since Scratch is written in Squeak Smalltalk it doesn't really match the git workflow. And of course, a patch would have to be Smalltalk code. If there are a number of experienced Smalltalkers wanting to contribute we might be able to find some way to arrange that.

stevensmi, 2015-10-29:
Thanks for the information. Sorry I could not get back on this earlier. As an experienced programmer I do not think contributing a Smalltalk patch will be two difficult. Not sure what others think! I spent a fair bit of time trying to find the 'nuscratch' source code but to no available. Would it be possible to at least upload a tarball. Thanks.

timrowledge, 2015-11-10:
The PiFace driver code was, it thought, actually setting the input pin pull-ups but the low-level code was missing out the pines value for the wiringPi  SPI handler. Fixed, along with correcting the input pin numbering map. Tests out on a Pi2/PiFace combo.

timrowledge, 2015-11-10:
Fixed in Scratch-tpr.292 & HardwarePeripherals-Core-tpr.20

closed 2015-11-10 by timrowledge