raspberrypi/scratch #157: enhancement request pwm--> servo

heppg, 2015-10-29:
Would be useful to have a flavor of pwm called 'servo'.
Value [0] servo --> [5% pulse width pwm, 1ms on 50Hz]
Value [100] servo --> [10% pulse width pwm, 2ms on 50Hz]

And usage of servo should set pwm frequency to 50Hz.

timrowledge, 2015-10-30:
That's pretty much what I was thinking after reading jhmns' message on the forum today. Servos need rather different wpm output than the 'ordinary' sort of thing like LEDs.

timrowledge, 2016-01-04:
This is actually a bit tricky with wiringPi.

We have a range of 1mS between -100% and +100% positions; if we want to be able to use those 200 'stops' then we need temporal resolution of 5uS which is simply impossible through wiringPi without using the 'delayMicrosecondsHard() call that times by a busy loop. That will generally completely mess up almost everything else, most especially on a single-core Pi.

timrowledge, 2016-02-20:
See #177 for an alternative approach

closed 2016-04-29 by timrowledge