raspberrypi/scratch #159: USB Scratchboard and Scratch 1.4 on Ras Pi

lcww1, 2015-11-08:
I've just noticed that current version of Scratch 1.4 on Raspbian no longer works with the USB Picoboard/Scratchboard (see https://www.sparkfun.com/products/11888 - this is the current board, and the version that I am using). This problem is present for both the latest Scratch version (20150915) and a preceding version from 14-Jan-15 that I happened to have installed on another Pi. 

The Picoboard itself is working perfectly under the latest version of Raspbian - the kernel drivers automatically load, and I can communicate with the board via python without any problems. 

I've also now found that a couple of older versions of Scratch 1.4 are also present on my Pis, namely those available under scratch.old (one of these has an image date of 20131203, whilst the other seems to be from 20111202) both of which work perfectly with the USB Picoboard. So, something seems to have changed between the 20131203 image and the 20150115 image to break the communication between Scratch and the USB Picoboard.

For now, I can just go with the scratch.old version, but it would be great to have the speed improvements of the latest version of Scratch for use with projects using the Picoboard.

timrowledge, 2015-11-08:
Thanks for testing it; I didn't even know they were still available. Well have to get one to try. All I can say right now is that nothing has knowingly been down to change anything related to it, so we must suspect differences between the ancient 2.8 era image and the newer 4.6 era.

lcww1, 2016-01-01:
Hi Tim!
I've found the answer to this problem, and I'm delighted to report that there's really no problem, and certainly no bug in your code! It seems that the default USB device for the ScratchBoard changed across Scratch versions at some point - on my Pi running the current Raspbian version the ScratchBoard/PicoBoard is attached to /dev/ttyUSB0 but Scratch needs to be told this in order to connect. The process is explained in this post on the Raspberry Pi Forum - 

https://www.raspberrypi.org/forums/viewtopic.php?p=871541#p871541

So sorry to have troubled you with this.......but nice to be able to get this working at last!

Best wishes, and a Happy New Year!

timrowledge, 2016-01-01:
Not a problem; it's quite likely that something changed under the covers in the fairly vast amount of time between 2.8 (Aug 2000) and 4.5 (Oct 2014).

I've just received a picoboard - nice bit of hardware, though a touch expensive seeming vs a Pi - and it does appear to work pretty well. I simply plugged it in, opened the 'Scratchboard Watcher', used that to 'select serial/USB port' and chose the only logical seeming option '/dev/ttyusb0'. Ta-da! Done.
Interestingly enough that's exactly what I had to do on my iMac running a copy of the very first Scratch 1.4 image from MIT, so I guess it does actually work as anticipated. Nice.

closed 2016-01-01 by timrowledge