raspberrypi/scratch #173: dropdown menu with many entries not useable

heppg, 2015-12-04:
Tried the midi-capabilities. Found that the menu with these many entries is not useable, when the menu list folds back on window edge. In the screen shot, please note that 1-33, 34-66, 67-99 open each one right of the last one, but 100-128 opens to the left. Found no way to move mouse into this last list.
When window is enlarged, so that the lists all open to the right, each list can be used.
[image]

timrowledge, 2015-12-17:
Interesting - what we are seeing here is that the submenus position nicely when they open to the right of the parent but a bit too far left when they hit the right end of the screen space. Thus as you move the pointer to aim at this last submenu you can't help but lose the focus (unless you move the mouse very fast!) on that menu.
Let's see exactly how it is calculating the position...

timrowledge, 2015-12-19:
Looks like the position used when we need to put the new menu to the left is being worked out as too far left; thus there is a gap (as seen above) and on some occasions that will trick the system into thinking the mouse has moved outside the menus and no selection is needed.
A simple fix that would place the last menu above (the one with 'gunshot' at the bottom) about 10 pixels to the right is being checked out.

timrowledge, 2015-12-29:
Fixed - included in the trunk Squeak release since it was an issue with main Morphic code.

closed 2015-12-29 by timrowledge