raspberrypi/scratch #178: enhancement: enable-flag for scratchboard-access

heppg, 2016-02-27:
For a physical computing workshop, someone in the Netherlands is planning to have arduino nano-boards connected to scratch. For this setup it is planned to use scratchClient.
Technically, the connection is a USB serial connection with FDDI chips provided by the nano..
The usage of nano boards is a low cost and breadboard friendly approach to work with raspberry Pi. It has 5V levels for the logic, provides accurate servo signals and provides some protection for the PI when wrong electrical connections are made. 

Unfortunately, scratch tries to look to /dev/USB0 from time to time and apparently changes baud rate for the connection. This disturbs serial communication to/from the nano. In addition to this, if someone makes a fault and enables one of the scratchBoard-sensor-variables, then a lot of wrong chars are sent to the USB.

Proposed solution: an optional flag 'scratchBoard=0' in .scratch.ini allows to disable the scratchBoard logic, especially the usage of /dev/USB0. Default value is '1' for backward compatibility. Best case the scratchBoard watcher and the associated variables are also supressed by this flag.

hansdejongehv, 2016-02-28:
I am the one in the Netherlands who was confronted with the following issue: Scratch (on Raspberry Pi, version of 2015-01-15) apparently continuously polls all /dev/ttyUSB* names and in that process sets the baudrate to 38400. Any program communicating over that port, e.g. scratchClient will get its communication disturbed. If the communication speed of the program is changed to 38400, it appears that every now and then trailing parts of the message up to end of line are disappearing, probably "eaten" by Scratch.

Hence the request to please create a way to configure Scratch such that this behaviour can be turned off.

Thanks very much.

timrowledge, 2016-02-28:
Yuck, that's nasty. I haven't ever really looked at that code so it may take a while to sort out. If you haven't opened the scratchboard morph to use it then it shouldn't be playing with the serial port at all!

hansdejongehv, 2016-02-28:
Thanks for your reaction. I do not exactly understand what you mean by "haven't opened the scratchboard morph". I am using version 1.4 on Raspberry Pi so I do not know whether it applies there as well. Anyway, I agree, I would not expect that Scratch starts scanning the ports unless I do something to tell that it should. I have not exactly found out which event lets it start, but I think definitely when I enable remote connections (which I should since I need to use Scratchclient). Note that it only has the described effects on the /dev/ttyUSB* ports, not on /dev/ttyACM* ports. So, as a bypass, at the moment I rename the ttyUSB* ports. But this is a bit clumsy. And it took me more than a day work to find out that Scratch was causing this, so that may happen to others as well.
Thanks for looking into this.

timrowledge, 2016-03-10:
The problem here is that the code has enough places where it expects to find a scratchboard connected that it is really hard to track down where it actually uses the serial port. Pretty much any time you use a sensor related block it will end up opening the serial port ago see if there is a board attached. It's all about trying to make it configure itself and avoid kids having to do anything fiddly.

About the only way I can think of to bypass this would be a global config option in the .scratch.ini file, which is a bit of a pain but should be able to cover the right cases.

I'd really like to remove the assumptions about the scratchboard and indeed the Lego We-Do connection and move them to the GPIOServer section but that would be a break with history.

heppg, 2016-03-10:
With a global option in .scratch.ini, the behavior could be:
- scratchboard=true (default)
  the scratchboard sensor variables are visible, the watcher can
  be enabled
  scratch checks from time to time if there is a serial
  connection on USB, tty...)
- scratchboard=false
  scratchboard variables are not visible, no watcher enabled.
  No attempts to look for a scratchboard.
- scratchboard=gpioserver,autoconnect
  the functionality moves to gpioserver. Gpioserver manages the
  visibility of sensor variables and allows the watcher.
  The watcher does not allow to select the connection (as
  this is managed by gpioserver, but is similiar to scratch
  1.4 on windows). This is the 'legacy' behavior.
- scratchboard=gpioserver,controlled
  gpioserver uses a new add-on-variable 'scratchboard' to
  enable scratchboard protocol. When this variable is not set,
  then this is like 'false' behavior.

gpioserver needs ability to allow multiple add-on boards in parallel.

These changes and an announcement that the default will change in a year
or so to 'gpioserver,autoconnect' could allow a painless migration to
new structure.

timrowledge, 2016-05-19:
Having just had the 'pleasure' of sorting out #193 I had to look into the serial port stuff a bit and I notice that the Scratch serial port handling code, when it attempts to open a port, 
a) uses open(bsdPath, O_RDWR | O_NOCTTY | O_NONBLOCK);
b) if the port opens, uses (ioctl(fDescr, TIOCEXCL) == -1) to 'lock' the port for exclusive access.
c) saves the port settings it finds active
d) restores them when closing
So what ought to be happening if you have some other device attached to a /dev/ttyUSB is that Scratch opens the port, sets the baud/RTS/DTR, sends a poll byte, reads whatever is returned, discovers there is no Scratchboard attached and return the prior port settings before closing. Obviously, that poll byte might cause some problems and if any returned data happens to look like Scratchboard data then Scratch will keep trying to use that port!

The bit that interests me is the ioctl(fDescr, TIOCEXCL part. If it is possible to claim exclusive access to a port, why doesn't this get done generally? It looks like the potential answer is that root level programs can open even a port locked in this manner, and since NuScratch has had to run as root for the GPIO access, maybe this has enabled crashing your party? The latest release of NuScratch does not run as root because of our change to pigpiod and thus, just possibly, your problem is solved as a side-effect?
GHP, does your code for opening the /dev/ttyUSB ports include claiming exclusive access? Could you add it?

timrowledge, 2016-10-31:
No news on this so I'm just going to close it for now

closed 2016-10-31 by timrowledge