raspberrypi/scratch #179: turn into new sprite: use very deep copy

heppg, 2016-02-29:
In backgrounds, when using function from context menu 'turn into new sprite', works üerfect when background is not having text. When text is included, there is an error message 'use very deep copy'.
The result is fine, text is available in sprite.

Problem: error mesage is not helpful.

timrowledge, 2016-02-29:
Ouch; sorry about that. Looks like I'm still using some old and deprecated code in there. I didn't even know you could turn a background into a new sprite...

heppg, 2016-02-29:
Didn't know either, but tried to reproduce some bugs from school. Kids
found that their sprites with text with 'umlaut' aka utf chars have been
displayed wrong. Did not find a path to this char error, but found this
one instead.

timrowledge, 2016-02-29:
As ought to have been obvious to me, the same issue exists when turning sprite costumes into new sprites. Happily, that means the fix solves both cases :-)
We simply need to actually use a #veryDeepCopy to copy the media item. Fixed ready for next release.

timrowledge, 2016-02-29:
Fixed in Scratch-tpr.323

closed 2016-02-29 by timrowledge