raspberrypi/scratch #182: Concatenation of output string

jrobinson-uk, 2016-03-18:
Really odd one:

User has written this script:
[image]

Which if you run (via desktop icon) and start the GPIO server fails to light a connected led.

When you change the script to start the GPIO server like this:
[image]
It works!

To complicate things if you then run scratch from a terminal, the issue is fixed and remains fixed for that file.

To reproduce:
- create script shown above
- attach an LED to GPIO 4

Original Thread:
https://www.raspberrypi.org/forums/viewtopic.php?f=49&t=140395&p=930235#p930235
https://www.raspberrypi.org/forums/viewtopic.php?f=49&t=139567&p=931988#p931988

timrowledge, 2016-03-21:
OK, the first potential issue here is that the illustration of the script at the top is wrong; the config should be of pin 14, not 4. It couldn't work otherwise, and in the original thread it is correct; so a likely future cause of confusion.

Using a corrected version, if I run it with the gpioserver not started, nothing happens as expected. Turn on the gpioserver and it works, no problem. The debugging tracer I have here in the lab shows that the commands being passed in are 
config14out
var: foo changed to: 14
gpio14on
gpio14off
as I'd expect. The original poster must have somehow left a space character in one or other parts of the 'join' operator block.

jrobinson-uk, 2016-03-21:
Tim,

despite the screenshot mistake (yes the original poster mixed 4 and 14) this is still an issue. @MarcScott and I used pin4 on 2 separate pi with the above script (with foo set at 4) and got the problem described.

When I'm back in the office Thursday I will video it :-)

timrowledge, 2016-03-21:
OK, but I can't replicate with pin 4 either! And the code for the concatenate is pretty unequivocal about not adding anything extraneous.

You might also add some explanation of what you mean by "if you then run scratch from a terminall, the issue is fixed and remains fixed for that file." because it makes no sense to me yet.

timrowledge, 2016-03-29:
Did you have anything further for this?

timrowledge, 2016-04-29:
Awaiting info...

timrowledge, 2016-07-07:
Still awaiting info..

timrowledge, 2016-10-31:
Since the gpioserver parsing has been changedforthe sept 16 release and would now handle a space etc in the stream I think we can close this

closed 2016-10-31 by timrowledge