raspberrypi/scratch #183: Scratch GPIO server crash

jrobinson-uk, 2016-03-18:
[image]
If you start the gpioserver in scratch (either from menu or script), stop it and then restart it. Scratch crashes

On exit wiringPi throws an error (attached)

timrowledge, 2016-03-18:
Yes, we've had a couple of reports of this  - see https://www.raspberrypi.org/forums/viewtopic.php?p=922976#p922976 etc and it is due to a change the policy of the wiringPi library.  Gordon-The-Author has been contacted, Simon Long knows and is expecting revised version to push out. 

This really should have been detected in pre-release testing an Pi Towers. I couldn't have found it in Scratch since there is no way of me knowing if a library changes at the last minute!

If there is a reasonably simple way to tell users to back off the wiringPi version (to 2.32 rather than 2.32) with some apt-get magic, we need to promulgate that ASAP. Oh, and it would be smart to make sure people in the teaching & learning resources know that reports of Scratch problems are likely to get seen *much* faster if they can post in the Scratch Forum.

timrowledge, 2016-03-21:
Simon, a reminder that we need to either get a revised version of wiringPi from Gordon to tell people to apt-get, or perhaps a command snippet to let them go back to the previous 2.31 version

winkleink, 2016-04-20:
I think I'm experiencing the same thing.

Raspberry Pi 3
Running 2016-03-18 with update/upgrade done.
I can start GPIO Server either from drop down or using 
broadcast gpioserveron

But if I then set a variable to an addon board
set AddOn to SenseHat
or
set AddOn to Pibrella

When I click the green flag to run the code Scratch closes down.

Appears to be the same issue so adding here to let you know the SenseHat is also affected 

Sample code that causes Scratch to close down included.

[gpioserveron.zip](https://github.com/raspberrypi/scratch/files/228796/gpioserveron.zip)

timrowledge, 2016-04-20:
Yeah, it's all the same cause. Next release will fix it; but worry not, I'm sure it will introduce its own new and exciting bugs. Err, 'issues' and 'opportunities'. Yep, that's what I meant.

timrowledge, 2016-04-29:
wiringPi library replaced by pigpen daemon. All low-level interfaces rewritten. Testing indicates good results and it is ready to go into next release barring showstopper bugs.

closed 2016-04-29 by timrowledge

winkleink, 2016-06-30:
I tested with SenseHat and latest Raspbian.  All worked perfectly. 
Thank you for fixing.

timrowledge, 2016-06-30:
You're welcome, and thank you for testing and reporting!