raspberrypi/scratch #184: Sprite scaling appears to ignore set centre point

timrowledge, 2016-03-31:
In NuScratch we appear to be ignoring the user set centre point of a sprite when scaling. See XXX for explanation and the forum-attached sb file for a nice demonstration.

timrowledge, 2016-03-31:
Ah, the WarpBlt>rotate... method has changed between 1997 & 2015. I'll have to work out how the centre position calculation has altered.

timrowledge, 2016-06-08:
After much distraction and confusion I finally got to look into this. The solution is decently simple - actually pay attention to the scaling factor within the calculation of the resized/rotated bitmap!
Fixed in Graphics-tpr.342 & Scratch-tpr.344 ready for next release.

closed 2016-06-08 by timrowledge

seanmcmanus, 2016-07-28:
Thanks for fixing this, Tim!