raspberrypi/scratch #185: No LICENSE-file in this repo

mobluse, 2016-04-05:
There is no LICENSE-file in this repo. Is this intentional? What is the license?

timrowledge, 2016-04-05:
Sorry, you've lost me already. What repo?

mobluse, 2016-04-06:
This repository: https://github.com/raspberrypi/scratch/
If this is an open source project it should have a license, such as GPL, MIT etc.

cymplecy, 2016-05-20:
The code (although based on open source) has been closed off

timrowledge, 2016-05-20:
That would be incorrect.

This is not a code repository but rather a place to stick artefacts to share in order to avoid trying to email giant files.

tobiasbp, 2016-05-20:
I too assumed the source for Scratch was available here. I see that it is not. I wonder why the source is not available? What is the license of the original code? Are there links that would be informative in this regard?

I have been evangelizing to my Scratch(On Pi)-using student about the wonders of open source under the wrong assumption that Scratch on Pi was open source. Ouch..

timrowledge, 2016-05-21:
See, the fact that you would assume that the source code would be here is an immediate indication that you don't understand Smalltalk. Aside from the prosaic fact that the code isn't in a state where I would wish to inflict it upon the outside world, Smalltalk is shared with Monticello packages kept in a suitable repository such as SqueakSource. Smalltalk is a live system, not a dead text in mundane ascii code files.

So no, the code is not 'closed off'. This is not a repository where the code would go. It does not, therefore, lack a license file. It is just a place where some bits get transferred and bug reports are collated.

closed 2016-07-07 by timrowledge