raspberrypi/scratch #186: Scratch doesn't recognise different instruments

seanmcmanus, 2016-04-05:
Scratch on the Raspberry Pi doesn't appear to play the different instruments. There are some small differences in note duration when different instruments are used, I think, but the tones sound alike.

I know this has been an issue for some time but I couldn't see it here so I've added it.

timrowledge, 2016-04-05:
I guess you're referring to the MIDI instruments in the 'play drum' and 'set instrument to' blocks, yes?
If so you need to do a quick install to load the MIDI sound libraries that provide the actual sounds - it was concluded that the fairly large library wasn't a sensible thing to include in the default Raspbian package.
The script you need to run is /usr/share/scratch/timidityinstall.sh - it doesn't seem to need 'sudo'. Do please add a comment her if it causes problems - and especially if it works ok! Outside confirmation is always good to see.

seanmcmanus, 2016-04-08:
Thank you, Tim! I didn't know that, but running that script has set it all up for me and the music works fine now. I ran the script with sudo (didn't read your note carefully enough) so I don't know if it works without, but it works fine with it.

closed 2016-04-14 by seanmcmanus

abee2, 2016-05-04:
@timrowledge If possible, could you please refer to the contribution of this function? https://www.raspberrypi.org/forums/viewtopic.php?f=77&t=88024&p=825177

timrowledge, 2016-05-05:
Hi Kazuhiro - I'm not sure what you are asking for here. I hope we've properly credited the original authors but if we haven't, tell me where you'd like it improved.

Oddly enough I was about to contact you to ask if anyone knows why the timidity service seems to keep taking more and more cpu time until it gets to 100%. I think it's a recent change but I can't be sure. It might be a bug in timidity, or perhaps we're not handling the MIDI stuff quite right in NuScratch.