raspberrypi/scratch #188: next-error message

heppg, 2016-04-11:
Some kids in class encounter this problem: an error popup is shown. Reason unknown.
[image]
One of the problems could be that the kids work for almost half a year with their code and a lot of load and save operations have been executed.

tobiasbp, 2016-04-11:
When does the error occur? On saving the document? What version are you
using?

timrowledge, 2016-04-11:
There are two things that can help me a lot if this ever happens
a) press that 'Debug' button! Unless something is truly terribly broken it will open a debugger (surprise!) and the top pane will list how it got into trouble. Curious kids (or adults) might even notice some of the details the resulting code shows. They might even get hooked on Smalltalk.
b) there *should* be a file aimed SqueakDebug.log (or something like that) in the home directory. Right now that ends up being the /root directory because of the annoying ( and soon to be fixed) requirement to run under sudo. Send me that file if at all possible.

The next release will have a fix to the problem that the error window is showing here, where it doesn't show the last few message sends, which would immediately give me better ideas as to what went wrong. My only decent guess from what we have here is that it was trying to save a file and something prevented the file opening (thus the sending of #next: to nil) and that might be because of the SD card running out of room?

heppg, 2016-04-14:
Here some first results from my last visit in school:
[SqueakDebug.log.txt](https://github.com/raspberrypi/scratch/files/219721/SqueakDebug.log.txt)

Version of scratch is "2015-09-15". Quite old, but we have extraordinary conditions that this course runs longer than half a year now.

Pressing 'debug' is done, but screenshots are still on machine (copy paste is (sometimes) a miracle). Takes some days till next lessons to get them.

Project is quite large, but works on my machine. So no more results from this.

timrowledge, 2016-04-14:
The debug log in clear - 
MessageNotUnderstood: UndefinedObject>>next:
6 November 2015 10:31:54.357 am

VM: unix - Smalltalk
Image: Squeak5.0 [latest update: #15113]

SecurityManager state:
Restricted: false
FileAccess: true
SocketAccess: true
Working Dir /usr/share/scratch
Trusted Dir /usr/share/scratch/secure
Untrusted Dir /home/pi/Scratch

UndefinedObject(Object)>>doesNotUnderstand: #next:
	Receiver: nil
	Arguments and temporary variables: 
		t1: 	next: 10000
		t2: 	MessageNotUnderstood: UndefinedObject>>next:
		t3: 	nil
	Receiver's instance variables: 
nil
MessageSocket>>sendStreamData
	Receiver: a MessageSocket
	Arguments and temporary variables: 
		t1: 	nil
		t2: 	nil
	Receiver's instance variables: 
		socket: 	a Socket[connected]
		inSizeBuf: 	#[0 0 0 0]
		inSizeIndex: 	1
		inBuf: 	nil
		inIndex: 	1
		outgoingMessages: 	an OrderedCollection('sensor-update "Name" "njhmnjmgbbbmbbmgk...etc...
		outBuf: 	nil
		outIndex: 	1
		lastActivityTime: 	3624258568

MessageSocket>>sendData
	Receiver: a MessageSocket
	Arguments and temporary variables: 
		t1: 	nil
		t2: 	nil
		t3: 	nil
	Receiver's instance variables: 
		socket: 	a Socket[connected]
		inSizeBuf: 	#[0 0 0 0]
		inSizeIndex: 	1
		inBuf: 	nil
		inIndex: 	1
		outgoingMessages: 	an OrderedCollection('sensor-update "Name" "njhmnjmgbbbmbbmgk...etc...
		outBuf: 	nil
		outIndex: 	1
		lastActivityTime: 	3624258568

[] in ScratchServer>>sendOutgoingCommands
	Receiver: a ScratchServer
	Arguments and temporary variables: 
		t1: 	a MessageSocket
		t2: 	'sensor-update "Name" "njhmnjmgbbbmbbmgkµħđŧj,gdhgdhbtd mzfrg,njzfrnj,uzrf"...etc...
	Receiver's instance variables: 
		stage: 	a ScratchStageMorph(3942326)
		userName: 	'Scratch'
		serverSocket: 	a Socket[waitingForConnection]
		incomingUDPSocket: 	a Socket[connected]
		peerSockets: 	an OrderedCollection(a MessageSocket)
		peerNames: 	an IdentityDictionary()
		sensors: 	a Dictionary()
		lastSentValues: 	a Dictionary()
[... 373 lines cut ...]
	Receiver's instance variables: 
		outerContext: 	MorphicProject>>spawnNewProcess
		startpc: 	53
		numArgs: 	0

--- The full stack ---
UndefinedObject(Object)>>doesNotUnderstand: #next:
MessageSocket>>sendStreamData
MessageSocket>>sendData
[] in ScratchServer>>sendOutgoingCommands
OrderedCollection>>do:
ScratchServer>>sendOutgoingCommands
ScratchServer>>stepServer
ScratchStageMorph>>stepServers
ScratchFrameMorph>>step
ScratchFrameMorph(Morph)>>stepAt:
StepMessage(MorphicAlarm)>>value:
WorldState>>runLocalStepMethodsIn:
WorldState>>runStepMethodsIn:
PasteUpMorph>>runStepMethods
WorldState>>doOneCycleNowFor:
WorldState>>doOneCycleFor:
PasteUpMorph>>doOneCycle
[] in MorphicProject>>spawnNewProcess
[] in BlockClosure>>newProcess

timrowledge, 2016-04-15:
OK, this is a bit tricky to work out. So far as I can see this is ancient MIT code having trouble (in fact stuff dated 2003!) and may be being triggered by the 'Name'->'njhmnjmgbbbmbbmgkµħđŧj,gdhgdhbtd mzf... part; that is presumably a unicode string because of the 'kµħđŧj,' characters. That in turn is what gets us from MessageSocket>>sendData to the MessageSocket>>sendStreamData method. 

What I see here is some messing with the buffering of socket data and evidently at some point all the data went out ok and we get to a curious bit of code that sets 'outbuf' to nil, and which would seem a reasonable cause for it being nil the next time around and thus causing the error. 

The code looks like - 
MessageSocket>>#sendData
sendData
	"Send some more data, if possible. Answer true if there is more data to be sent. This method should be called frequently by the client. It does nothing if there are no messages to send, or if the socket is not connected or ready to accept data. If outBuf is not a bytes object (i.e., a ByteArray or String) then assume it is a positionable stream and let sendStreamData send it."

	| buf n msg |
 	(socket notNil and: [socket isConnected]) ifFalse: [^ false].
	socket waitForSendDoneFor: self standardTimeout.

	outBuf ifNil: [  "not currently sending"
		outgoingMessages size = 0 ifTrue: [^ false].  "no more message to send"
		socket sendDone ifFalse: [^ true].  "still sending last data"
		outgoingMessages first class isBytes ifFalse: [^ self sendStreamData].
		buf := WriteStream on: (ByteArray new: 2000).
		[outgoingMessages size > 0] whileTrue: [
			msg := outgoingMessages removeFirst squeakToUtf8 asByteArray "it really ought to be practical to convert directly to a byte array".
			buf uint32: msg size.
			buf nextPutAll: msg ].
		*outBuf := buf contents.*
		outIndex := 1].

	outBuf ifNotNil: [  "currently sending outBuf"
		outBuf class isBytes ifFalse: [^ self sendStreamData].
		n := socket sendDataNoWait: outBuf startIndex: outIndex.
		n > 0 ifTrue: [
			lastActivityTime := Time totalSeconds.
			outIndex := outIndex + n.
			outIndex > outBuf size ifTrue: [outBuf := nil]]].

	^ true
The bolded line *outBuf := buf contents.* is the only place where 'outbuf' is set to anything non-nil, and it is always set to the contents of a stream made from the outgoing message text (which may be simple ascii bytes or a 32bit wide unicode string converted to utf8 encoding as appears to be the case here) with a small header to specify how long it is.
However, if the 'outgoingMessages first' - i.e. the first item in the list of outgoing stuff - is not a byte string (which it isn't, here) then we can get to - 
MessageSocket>>#sendStreamData
sendStreamData
	"Private! Used by sendData only when outBuf is a positionable stream. Send some data from outBuf, if possible. Set outBuf to nil when the entire stream has been sent. Always answer true. "

	| tempBuf n |
	tempBuf := outBuf next: 10000.  "read up to 10000 bytes from the stream"
	outBuf skip: tempBuf size negated.
	n := socket sendDataNoWait: tempBuf startIndex: 1.
	n > 0 ifTrue: [
		lastActivityTime := Time totalSeconds.
		outBuf skip: n.
		outBuf atEnd ifTrue: [outBuf := nil]]. "finished sending outbuf"

	^ true
 - where we immediately send #next: to outbuf, which we know is nil and kaboom; a veritable Earth-shattering Kaboom.

Tracing this all back a bit we see that the only input point is within ScratchServer>>#sendOutgoingCommands, which derives the data to be sent from ScratchServer>>variableUpdateMessage. Old Scratch kept any non-Latin1 related strings as a weird class 'UTF8' which got converted in odd places and somewhat confusing times; NuScratch leaves strings in either byte or wide form as much as possible; I evidently missed a place where a wide string might need converting to utf8 encoding before sending via a socket.
Since the old Scratch 'UTF*' class is a subclass of String and thus 'blahblah class isBytes' is true, we would never actually get to run #sendStreamData and thus never have this particular problem.

Sigh. Now to fix it.

timrowledge, 2016-04-15:
The problem is that old Scratch assumed all strongish things were UTF8 class objects, which meant them being byte objects. So the line 
   ` outgoingMessages first class isBytes ifFalse: [^ self sendStreamData].`
can be changed to 
    `outgoingMessages first isString ifFalse: [^ self sendStreamData].`
to handle both wide and byte Strings. This  should solve our problem cleanly with minimal changes. It is probably the case that the entire thing could be cleaned up more since it seems nothing can actually add non-String data to the outgoingMessages list and so the sendStreamData method is almost certainly redundant.

Fix prepped for next release.

GHP, as a test, you might like to make that change in your working image.

timrowledge, 2016-10-13:
This went out in Scratch-tpr.334 and the following release

closed 2016-10-13 by timrowledge