raspberrypi/scratch #192: The range of GPIO PWM is 0-100, not 0-1024.

abee2, 2016-05-04:
The [document ](https://www.raspberrypi.org/documentation/usage/scratch/gpio/README.md) says as following.

`gpio + pin number + pwm + [ 0..1024 ] (we can use the software-driven PWM output facility)`

However, the pin that is set more than 100 always outputs 3.3 V.
It seems the correct range is 0-100.
Is it a bug or an error of the document?

timrowledge, 2016-05-05:
You are right; it's a bug in both the code and the document, making it even worse.

abee2, 2016-05-05:
Thanks. But, what is the correct range?
I'm writing a book about the GPIO server now. So I'd like to know that. I'm sorry about my private reasons.

timrowledge, 2016-05-05:
Ouch, sorry about that. 

OK, right now in the jan 2016 release I can see a bug in my use of the wiringPi softPWM mode - I didn't notice that the default value range is set to 100 and not the 1024 that the hardware pwm assumes. Such is the joy of peering at other people's C code libraries.

The version about to be  released uses pigpio instead (to provide ultrasonic and serve support etc) and I think I have found a problem in that the default pwm range is 256, not 1024. This is also going to be a problem for the assorted motor driver boards that work off the pwm system. I can see how to fix it but I think it is too late to get those fixes into the next release.

For your book, unless it is just about to be published, I think we can make sure the real system matches the book by the time it reaches the printer. I would suggest documenting the pwm as having 0-1024 range (actually, shouldn't that be 0-1023? Well, that's something to change the doc a little for) and the motor command as using -100...+100. In fact, since writing a book is very like writing a specification, please, please tell me everything you find that is difficult to understand, not consistent, could be better, is missing, or in any other way not good. Wherever possible I will make use of any such comments to clean up the gpioserver. 
As I think you might know, my email is simply tim at rowledge dot org

timrowledge, 2016-05-25:
Fixed ready for next release; motors are set to have 0..100 rang and general pwm 0..1023

closed 2016-05-25 by timrowledge