raspberrypi/scratch #193: Port scanner for sensor board does not work correctly.

abee2, 2016-05-15:
It cannot find /dev/ttyUSB*. Because it detects /dev/ttyAMA0 first.
I think SensorBoardMorph>>portNames should reject the port name that has 'ama'.

timrowledge, 2016-05-15:
I don't really know a lot about unix serial port naming conventions so this is a bit hard to get into. The sensor board asks for the available ttyusb and ttyAMA ports and then defaults to the first one in the list. You can easily choose any alternative port with the menu 'select serial/USB port' option and it certainly works on my Pi's.
Let's see if I can find any understandable info about this...

abee2, 2016-05-16:
/dev/ttyAMA0 is mapped to the serial console on GPIO as default. So it is not used for the sensor board. The scratch.old ignores it, and port scan works fine.
I know I can open the port with the menu. However, sometimes it makes confuse my pupils. They need to open the port manually when they open their project that is using sensor board.
I hope the port scan works on NuScratch.

timrowledge, 2016-05-16:
This has some very strange history. The old, original serial port code does not list the ttyAMA* port(s) at all, which would explain why the Scratch code did not need to exclude it for the sensor board. However, according to the change log for the relevant file on squeakvm.org, I added detection of the ttyAMA port on 22 nov '15 as part of checking in the fixes for the MIDI sound output. I can't find any sort of note about why that was done, and it isn't mentioned in the MIDI changes by Manabu Sugiura - so what on earth was it for? 
Eventually after much searching of records, I see it was added actually to support the PiLite HAT, which uses the internal serial port - ttyAMA. So I think that as we know the sensor board can only be attached via usb we can just exclude the AMA port.

timrowledge, 2016-05-16:
Excluding the AMA port works ok, but does not make the sensor board connect to it by default.
The code to open the port works perfectly but something somewhere immediately disconnects it and sets the port back to nil.

timrowledge, 2016-05-16:
On my iMac using the very old original MIT Scratch, the sensor board does not manage to auto-start either, requiring the user to choose the right serial port.

timrowledge, 2016-05-17:
And just to make life interesting, an ancient copy of Raspbian on a Pi B, with original Scratch *mostly* automatically connects to the sensor board but not always. Sigh.

timrowledge, 2016-05-17:
So far as I can work out today the state machine is demented.

timrowledge, 2016-05-18:
Well this was fun, for interesting and unusual definitions of the word 'fun'...
After much tracing, tracking and printf'ing I finally discovered that one of the methods deep in the state machine was passing the sensor board serial port a poll byte of value 1 by actually using 'port nextPutAll: #(1)' -which will actually send an Integer array, not a byte. In another method we were using the correct constant byte array syntax - #[1]. Change to using that and the sensor board starts working as soon as it is noticed by Scratch. I suspect the oldest version was working occasionally by fluke of not totally reliable serial transmission.
That was a fun way to spend three days 😬

timrowledge, 2016-05-18:
See also #178

timrowledge, 2016-05-19:
Fixed in Scratch-tpr.340 for next release

closed 2016-05-19 by timrowledge