raspberrypi/scratch #198: scratch and pibrella

thelegoman, 2016-07-04:
Hi. I have a problem getting scratch to accept the pibrella. As soon as its named as the add on, scratch closes down
Any thoughts? Thanks

timrowledge, 2016-07-04:
Please add some more details about which version you are using. All you need to do is start Scratch and look at the title of the main window.
The most likely explanation is that you are using a version that shows 'Scratch 1.4(Linux) of 2016-01-15'. Unfortunately when Pi Towers released this version of Raspbian the 'wiringPi' library had been altered without anyone telling me and that alteration made a lot of the gpio server code have problems.

You need to update your system to the latest. You *should* find that `sudo apt-get update` followed by `sudo apt-get upgrade` does the trick. If you feel the need to try upgrading only Scratch then do remember that you will want 'nuscratch' not 'scratch' since that refers to the old, slow, system!

I'll note that the Pibrella seems to cause some problems even on the latest system, depending on as-yet unknown conditions. Some users have reported (even outside Scratch) that pressing the big red button can seem to lock up the Pi, for example.

timrowledge, 2016-10-05:
I'm going to close this since the Pibrella handling and base library are quite different now

closed 2016-10-05 by timrowledge