raspberrypi/scratch #209: 20180824, pen down and save project

heppg, 2016-08-30:
... draws funny lines on stage. In the sample picture, there are two sprites on stage. Both have pen down, pen size set to quite large. Then project is saved and the sprites draw a line.
[image]

timrowledge, 2016-10-13:
Ah, I see the problem.

When changing the stage co-ordinate mapping from {insert rant about ridiculous ad hoc value mangling here} to a proper transform morph wrapper, that caused the pen position updating in #writeScratchProject to mis-behave. We simply don't need to move the unwrapped stage position to 0@0 - the reload & re-wrap code will handle any position.

timrowledge, 2016-10-13:
Oddly, the pen down state is carefully *not* saved with the project - checked this with the original to be sure. Can't think why that choice was made.

Fixed in NuScratch-tpr.371

closed 2016-10-13 by timrowledge