raspberrypi/scratch #210: 20160824, numeric input fields do not accept komma as dec separator

heppg, 2016-08-30:
For numeric fields, e.g. in '+'-Operator, it is possible to enter decimal values like '3.1415'. The field does not accept a komma as separator, so not possible is '3,1415'. 
Legacy scratch 1.4 accepts komma for decimals and converts these to dot on enter.

Remark: germany is a comma numeric world. Makes live easier for kids to stay in their known world..,,..

timrowledge, 2016-09-06:
Yup, that should be fixed. It's one of those things that I (as a UK born english speaker) simply wouldn't spot because we virtually never  come across it.

timrowledge, 2016-10-17:
The string parsing to make a number from '3,1416' etc seems to work perfectly well. 
Rather oddly the main code I see to read strings and convert to numbers when needed (ScriptableScratchMorph>>interpretStringAsNumberIfPossible:) carefully checks for a dot and excludes a comma - the same as the code in legacy Scratch. We can see this by dragging a > operator block into the script editor and entering 4 on one side and 3,5 on the other. If you click on it the result is false, which looks good until you change it to 3 < 3,5 which is also false :-) Again, this works the same in both versions.
An example project would help a lot here.

heppg, 2016-10-18:
Wherever a round entry box is available, there can be numeric values.
Examples are the plus operator, rounding or the other numeric operators.

In the test, I typed a komma into these round entry fields and found
that this was not accepted as a valid entry by the widget. So the chars
typed in sequence '3' ',' '1' '4' have been recorded as '314' instead of
'3,14'. It accepts [0-9], dot and minus.
The problem is that the widget does not accept kommas.

Then, on 'commit' which is either 'enter' or e.g. start button, the
input value is converted to '3.14'. This is fine.

As one conclusion, I can't provide a script, as the data in the system
are always 'dot-ified'.

timrowledge, 2016-10-19:
Interesting and a bit odd. I didn't notice that the rounded expressionargmorphs are set to be numbers and the squared ones are not...
The StringFieldMorph used to provide the text and editing can be set to numeric mode and then only allows typing of -0123456789., - which is especially amusing since the code intended to convert strings to numbers when possible  (asNumberNoError) carefully  makes sure to allow an initial + as well. And handles exponents. But then interpretStringAsNumberIfPossible: only allows -0123456789.

What a combination of confusion.

Good news so far - I have a new number parser that copes with . or , and is several times faster. Since the constant converting from string to number tends to take 10-20% of time for things like PacMan this may make a useful difference!

timrowledge, 2016-10-20:
The usage of string to number conversion is very erratic in  the legacy scratch code and confusing enough that I'm going out any serious revision to one side for now. Solving the comma-for-decimal-point issue is simple enough and will improve things for users.

timrowledge, 2016-10-20:
The simplest fix for the original problem is to add a check for $, to StringFieldMorph>>insertCharacter: (which was there in a different form in legacy scratch). The round entry boxes (aka ExpressionArgMorph with isNumeric -> true) now accept 3,5 and convert it to 3.5 as before.

timrowledge, 2016-10-20:
Fixed in NuScratch-tpr.374

closed 2016-10-20 by timrowledge