raspberrypi/scratch #212: 20160824, xposition of variable shows tenth of pixel

heppg, 2016-08-30:
in legacy, the xposition or yposition variable watchers display interger only. In current scratch, there are decimal places shown.

timrowledge, 2016-09-06:
So they do. Shouldn't!

timrowledge, 2016-09-11:
I'm confused now. In original Scratch image I see one decimal placement of the time for the x/y position of a sprite when looking at the 'x position' or 'y position' blocks, which is just what I see in the latest NuScratch. I can make a nasty mess in both version if I do
set x pos -32
change x by 0.4
click on a 'x position' block and get a typical floating point ridiculous -31.499999999999 type value. Sigh. But at least it is both versions!

For the position watcher in the script editor area I only see integer values in both versions.

For the cursor x/y position below the stage I see integer in original scratch  and... no moving values in NuScratch. Oh rats.

timrowledge, 2016-09-11:
...And yet in the released version of the image the cursor tracking is just fine. I'm sure there is a good reason but I don't remember it just now...

timrowledge, 2016-09-13:
GHP, if you can grab a screenshot of where the problem is it would help, I think.