raspberrypi/scratch #213: 20160824 no border around slider min, max entry fields

heppg, 2016-08-30:
The dialog shows no border around entry field. In legacy scratch, there is a broad border.

timrowledge, 2016-10-13:
So there is. Or isn't, rather.

timrowledge, 2016-10-13:
I'd never even seen these dialogues before!

timrowledge, 2016-10-13:
Some changes in the way borders are handled mean a slight fix is needed to make a border visible here - simple enough.

timrowledge, 2016-10-14:
Well actually not at all simple to debug since setting the border colour to #inset - in order to get a border - does not actually make the border until the next time the system needs to know how any border might affect the bounds, or draw itself etc. So it gets to be very interesting to try to trap this in action.
It turns out that SimpleBorder>initialise was changed recently to set both baseCOlor and color to transparent - which along with a recent change to #trackColorFrom: which is used to set the border colour to suit its owner each time it is redrawn  - meant all borders were transparent unless explicitly set to a color.

timrowledge, 2016-10-17:
In order to avoid error 33 this is worked around in the fix NuScratch-tpr.372

The 'better' fix is committed for the next squeak release

closed 2016-10-17 by timrowledge