raspberrypi/scratch #215: 20160824 command line start, presentation mode

heppg, 2016-09-03:
Command line switch 'presentation' does not work.

Starting a program with program-path is ok
example: scratch 'Documents/Scratch Programs/test.sb'

Adding the parameter 'presentation' does not work and causing 'file not found errors in scratch.
example: scratch presentation 'Documents/Scratch Programs/test.sb'

timrowledge, 2016-10-17:
Oh, the joys of bash scripts.

/usr/bin/scratch simply wouldn't allow a single word parameter, having been written expecting absolutely every option to require a sort of {name} {value} pair.

After far too much banging my head against the desk I finally managed to add handling of single word  options as requested. You can now do
**scratch fullscreen --document /my/dir/to/scratch/project.sb**
or
**scratch presentation /my/dir/to/scratch/project.sb**
for example. Even 
**scratch fullscreen**
seems to work now

timrowledge, 2016-10-17:
Gerhard, would you like to quickly  try out the attached replacement /usr/bin/scratch shell script?

[scratch.txt](https://github.com/raspberrypi/scratch/files/534793/scratch.txt)
Obviously you'll have to rename it back to just 'scratch' and set the execute permission.

timrowledge, 2016-10-22:
Included in package for next release

closed 2016-10-22 by timrowledge