raspberrypi/scratch #216: 20160824 command line start, terminate program by window 'X'

heppg, 2016-09-03:
Start a program on command line
example: scratch 'Documents/Scratch Programs/test.sb'
Terminate program by click on X in window title.
[image]

Same setup, but terminate program by 'file/quit' is perfect.

timrowledge, 2016-09-06:
Hmm, interesting. What happens here is that a WindowEvent is passed in by the VM to handle the X close and it somehow ends up being queried for 'position' by the Scratch stage. Quite how that happens is a mystery right now. Let's take a look.

timrowledge, 2016-09-06:
Oh, of course. The WindowEvent is raised by the VM as a result of clicking on the window's X button, which is caught by the windowEventHandler I install as part of the release build. This passes the WindowEvent on to ScratchFrameMorph, which ends up using ScratchCloseDialog>getUserResponse. As part of the somewhat tacky modal dialogue handling in Morphic the various submorphs in the Scratch application get to do their stepping which includes the little mouse position stringmorphs wanting to display the latest mouse x/y.

This is a side-effect of some changes to event handling in Squeak 5.1 that means we should be using 'lastEvent' in ScriptableScratchMorph>>mouseX/Y instead of 'currentEvent'. Damn.

timrowledge, 2016-09-13:
Fixed in NuScratch-tpr.363
Replaced more (mis)uses of currentEvent by either currentHand lastEvent or by pushing the actual event down the stack.

closed 2016-09-13 by timrowledge