raspberrypi/scratch #225: "stop all sounds" is not working

heppg, 2016-10-07:
play a sound (e.g. 'cave' from 'music loops')
after a short while, execute 'stop all sounds'
Problem: the sound is not stopped
When all scripts are stopped by red button, the sound is stopped too.

timrowledge, 2016-10-07:
That's interestingly odd. It isn't related to the 'wrong-vm-released' problem #224 so I wonder...

heppg, 2016-10-07:
One of the kids in school found this on the may-16-release and I
doublechecked on current release.

timrowledge, 2016-10-17:
That was interestingly weird.

In old Scratch we used ScratchStageMorph>>stopAllSounds 
SoundPlayer shutDown.
self mididAllNotesOff

and that relied - wrongly - on the #shutDown method that is intended to stop sounds and clean up any external resources before quitting. In more recent Squeak systems a different path to releasing resources is used and the shutDown did nothing at all.

Simply using the more precise SoundPlayer stopPlayingAll makes the block work correctly.

timrowledge, 2016-10-17:
Fixed in NuScratch-tpr.373

closed 2016-10-17 by timrowledge