raspberrypi/scratch #226: NuScratch20160915 does not detect /dev/ttyAMA0 (GPIO) as a sensor board

neuralassembly, 2016-10-17:
I was using some sensors via /dev/ttyAMA0 (serial port on GPIO) with NuScratch02052016.
However, it is no longer detected by the latest version of Scratch.
Is it possible to support /dev/ttyAMA0 again?

I think that this problem might relate to the fix of #193,
i.e., about the automatic detection of /dev/ttyUSB0.

The port /dev/ttyAMA0 always exists even when the sensor is not connected to GPIO,
so, in order to keep the automatic detection of /dev/ttyUSB0 fine,
I think that the priority of /dev/ttyUSB0 for the port scan
should be higher than that of /dev/ttyAMA0.

timrowledge, 2016-10-17:
What sensors are you using and how are you using them?

Right now I can't find any changes to serial port stuff that seem like they might affect you. The #193 stuff was specifically about the scratchboard which ought not cause you any problems - it literally just excludes the AMA port(s) from the list offered for the scratch board.

Explanation and example code would be very helpful here - quickly if possible since I'm wrapping up a new release to fix a serious key-press problem.

neuralassembly, 2016-10-18:
To a PIC micro controller, I wrote a program that reads analog sensor and sends its values to Scratch via /dev/ttyAMA0
based on the "[Scratch Board Serial Communication Protocol](http://swikis.ddo.jp/WorldStethoscope/uploads/54/ScratchBoard_Tech_InfoR2.pdf)".

Then I made a circuit as shown in [this figure](https://1.bp.blogspot.com/-ySL-xZ-vjKE/WAWipcJS04I/AAAAAAAAE2c/P8YFUBW1_eEnyg8hCiqNi93AhR6LzuIWwCLcB/s1600/SensorCircuit.png).

Even when we do not have the above circuit, we can check whether the above circuit works or not.
When I try to open the port, NuScratch02052016 showed /dev/ttyAMA0 as shown in [this figure (a)](https://1.bp.blogspot.com/-qSL55j6X-Sw/WAWipIPRnOI/AAAAAAAAE2Y/Ac8OFU7pT2k9Cys_cDJAZ8OSz_6mvdtUgCLcB/s1600/SensorBoardExp.png).
When I choose it, Scratch starts the communication with PIC microcontroller.

However, the latest Scratch tells that there is no serial port as shown in [this figure (b)](https://1.bp.blogspot.com/-qSL55j6X-Sw/WAWipIPRnOI/AAAAAAAAE2Y/Ac8OFU7pT2k9Cys_cDJAZ8OSz_6mvdtUgCLcB/s1600/SensorBoardExp.png).
Therefore, Scratch cannot communicate with the PIC microcontroller.

timrowledge, 2016-10-20:
I'm impressed - custom hardware for scratch. Cool stuff.

Thank you for the pictures illustrating the problem. I wish more bug reports were as clear. Initially I couldn't work out how you were connecting and had no idea people might make their own hardware like this. Now I know that this is a simple case of excluding serial port names too aggressively and that's easy to fix.

I've changed SensorBoardMorph>>portNames to select any 'usb' ports first and then any 'ama' ports and present them in that order in the menu.
Testing with my scratchboard shows that it still works with this change and the menu now lists
/dev/tty/USB0
/dev/tty/AMA0

timrowledge, 2016-10-20:
Fixed in NuScratch-tpr.375

closed 2016-10-20 by timrowledge