raspberrypi/scratch #227: Instructions error in Examples

DarrenTownsend, 2016-10-24:
In the two ultrasonic examples, the wiring instructions say to connect Trig to gpio26 and Echo to gpio16, but these are reversed in the scripts.

timrowledge, 2016-10-24:
Damn! All hail Tpyos, Queen of spling misteaks.

Well spotted. I didn't think anyone ever read my notes...

DarrenTownsend, 2016-10-24:
I've been away from it for a while (THB partly due to our discussion regarding servos), but having another look now I've started an after-school club. I may actually propose some additions & edits to the docs, once I figure out how GitHub works!

timrowledge, 2016-10-31:
If the revised project files are small enough to just email, feel free to send them to tim at rowledge dot org

DarrenTownsend, 2016-11-01:
Will do, Tim.

timrowledge, 2016-11-29:
OK, I've fixed the files at my end; happily the Fritzing graphic I did was already correct.
It'll go out in the next update.

closed 2016-12-13 by timrowledge