raspberrypi/scratch #228: 2016-09-15 key presses not being captured reliably

winkleink, 2016-10-24:
Just did an update/upgrade on Pixel.
Ran scratch as saw NuScratch was updated.
Now keypresses are not being captured correctly.

Playing Pacman for Scratch  it starts OK and then stops responding to key presses. Pac Man stops changing direction on keypress
In space invaders it 'sticks' either left or right.
In Asteroids the ships keeps spinning in one direction or keeps firing.

Doesn't appear to be a problem in 3 Trampoline so maybe it's related to the complexity of the code. 

Running on a Pi3.

If I [alt]-[tab] out of scratch to a leafpad document the keyboard works perfectly.

timrowledge, 2016-10-25:
This is a fresh update from pre-Pixel to Pixel, right? Assuming so, this is the problem referred to in #224 and discussed in https://www.raspberrypi.org/forums/viewtopic.php?p=1047780#p1047780 etc
There's a test package you can try (if you're brave/foolish enough) mentioned further down the forum thread.

If this *isn't * that, make sure to let me know!

winkleink, 2016-10-25:
Nope. 
Was Pixel already from fresh installl. Can't be doing with sister upgrades. 
Did Pixelupgrade yesterday. 

Didn't test Scratch on Pixel before so may not be related to yesterday's update but something that was there before.  

Also, in case useful. I'm using a USB wires keyboard. Not Bluetooth.

timrowledge, 2016-10-31:
OK, then this is 99.9% likely the known and (hopefully) fixed issue #224 New release coming soonish with Raspbian maintenance release.

closed 2016-12-12 by timrowledge