raspberrypi/scratch #229: script, context menu allows text input

heppg, 2016-10-29:
Context menu 'right mouse click menu' collects keyboard inputs and display as extra line.
[image]
Example is from a context on a script fragment. But works in palette blocks too, and also on stage.
 
Expected behavior: keyboard inputs are ignored in context menu.

Issue was spotted by two kids in school class.

timrowledge, 2016-10-29:
Umm. I have no idea why a menu would ever do that; quite bizarre. I wonder where that code came from...

timrowledge, 2016-10-29:
Oh, that's where. At some point somebody decided that menus would be more helpful if you could tap some characters that are part of the menu item you are looking for and only the entries matching would be highlighted and selectable. It's actually quite a clever idea when you have long menus; for example open the languages menu (with the globe icon in the menu bar) and type 'de'. You should see that Deutsch and Nederlands get highlighted and al the others are greyed out. Neat, eh?

I don't think it's a particularly good idea in the short menus used by Scratch. I'll see where it gets turned off.

timrowledge, 2016-12-08:
A simple preference change sorts this out. Fixed in NuScratch-tpr.379

closed 2016-12-08 by timrowledge