raspberrypi/scratch #230: 2016-10-21, drag variable on stage shows black edges

heppg, 2016-12-03:
When dragging variables on stage, the edges are black. Should be transparent ?
[image]

timrowledge, 2016-12-03:
That is ... bizarre. It doesn't happen in the development image, but does if I run the release/build script to make a release version. I'm baffled right now. How?....

timrowledge, 2016-12-06:
Some rather desperate flailing around shows that this odd effect is caused by changing the default corner radius for rounding morphs; which was done to make the pull-down menus from the menubar look better. Oddly enough it is only the value 8 that seems to cause this - 7 or 9 or 10 are all fine!

timrowledge, 2016-12-06:
OK, definitely not a bug I caused; it works the same (wrong) way on Mac in plain Squeak 5.1. At least that means I'm not going nuts.

timrowledge, 2016-12-08:
An ancient bit of hack that used to improve performance when routing morph corners was looking for 48 pixels of transparent bitmap. This is what a rounding of 8 causes; so using a default radius of 8 gets caught . The corresponding code that would then do 'fast rounding' is no longer in the system. Sigh.

So, although a fix is easy enough for future systems, I need a solution within the 5.1 image. Right now the best thing is to changes the default radius to 7 instead.

timrowledge, 2016-12-08:
Fixed in NuScratch-tpr.378

closed 2016-12-08 by timrowledge