raspberrypi/scratch #237: Single script block not working

8119436437056971, 2021-10-02:
In a script a single block did not work. See attached image for the script.
One of the goto_x_y-blocks did not execute, the turtle graphic should be a set of nested squares but there are only triangle shown.
[image]
The image is edited to consume less space.

Isolating the affected block (remove it from the script as a single block) and execute it did not succeed.

Saving script and run it in a fresh started nuscratch instance succeeded.

The script was produced by a kid in a school lesson.

timrowledge, 2021-10-02:
Unfortunately I can't find any way  to replicate this, which makes it impossible to debug.
The only faintly possible idea I have is that somehow the "moveto x: 0 y: a" block wasn't quite dropped in the right place and didn't get properly linked into the stack. That would make it unlikely that the project would write out and then run correctly later, so not a very good idea.

> On 2021-10-02, at 4:43 AM, 8119436437056971 ***@***.***> wrote:
> 
> 
> In a script a single block did not work. See attached image for the script.
> One of the goto_x_y-blocks did not execute, the turtle graphic should be a set of nested squares but there are only triangle shown.
> 
> The image is edited to consume less space.
> 
> Isolating the affected block (remove it from the script as a single block) and execute it did not succeed.
> 
> Saving script and run it in a fresh started nuscratch instance succeeded.
> 
> The script was produced by a kid in a school lesson.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Borrow money from pessimists--they don't expect it back.

8119436437056971, 2021-10-02:
Unfortunately I do not know the history of this instance of scratch at this point, so no clue what happened before. The kid was happy when it worked after store/reload.

closed 2021-10-02 by 8119436437056971

timrowledge, 2021-10-02:
> On 2021-10-02, at 2:05 PM, 8119436437056971 ***@***.***> wrote:
> 
> 
> Unfortunately I do not know the history of this instance of scratch at this point, so no clue what happened before. The kid was happy when it worked after store/reload.

Well, that's the bit that matters.

tim
--
tim Rowledge; ***@***.***; http://www.rowledge.org/tim
Fractured Idiom:- L'ETAT, C'EST MOE - All the world's a stooge